Makerfabs/Makerfabs-ESP32-UWB-DW3000 #3: Multiple Anchors

SimonSchirber, 2022-11-15:
Is there a demo that has an initiator pining multiple responders/anchors? Also what is the best way to tune this, I tried changing the delay constants but it seems to have non-linear error signifying you cant just tune it with a constant change in the delays.

petex8989, 2023-03-23:
Were you able to figure out how to use multiple anchors?

LowellSeal, 2023-08-09:
Simon, did you ever get anything working for TWR with multiple anchors?

petex8989, 2023-08-10:
I was able to find a way to connect multiple anchors, you can check it out in my project on GitHub
https://github.com/petex8989/Indoor_Positioning_DW3000

kyuhyong, 2023-08-21:
Hi @SimonSchirber. Thanks for your post. 
I tried to upload the code you posted but could not get it to work. Can you explain a little bit about the setup?
I'm also trying to set 2 anchors stationary on a platform and 1 tag is moving and detect the location of the tag on a stationary platform. Is this setup possible?