Makerfabs/Makerfabs-ESP32-UWB-DW3000 #4: Ranging using 9 channels

MitsuyoshiHori, 2022-11-24:
As We mentioned in my e-mail, when We changed the settings from 5ch to 9ch in the following sample, "CONFIG FAILED" due to "PLL LOCKING ERROR" occurred and it did not work properly.

[sample]
Dw3000\examples\ex_06a_ss_twr_initiator\ex_06a_ss_twr_initiator.ino
Dw3000\examples\ex_06b_ss_twr_responder\ex_06b_ss_twr_responder.ino

[Modified part]
static  dwt_config_t config = {
        5 , --（Changes） --> 9 ,

[operation log]
---------------------
Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
setup() run 9ch...
SS TWR INIT v1.0
DEVICE ID: deca0302
XTRIM OTP READ FAIL
PLL LOCKING ERROR
CONFIG FAILED
---------------------

However, as a result of putting the following serial output in the dw3000_device_api.cpp file, it works even with 9 channels, strangely enough.
We will share this information in the hope of further improvement.

---------------------
    ///////////////////////
    // RF
    if (chan == 9)
    {
        UART_puts("RF ch9 \r\n");   // Add for debug
        // Setup TX analog for ch9
        dwt_write32bitoffsetreg(TX_CTRL_HI_ID, 0, RF_TXCTRL_CH9);
        dwt_write16bitoffsetreg(PLL_CFG_ID, 0, RF_PLL_CFG_CH9);
        // Setup RX analog for ch9
        dwt_write32bitoffsetreg(RX_CTRL_HI_ID, 0, RF_RXCTRL_CH9);
    }
    else
    { 
        UART_puts("RF ch5 \r\n");   // Add for debug
        // Setup TX analog for ch5
        dwt_write32bitoffsetreg(TX_CTRL_HI_ID, 0, RF_TXCTRL_CH5);
    

    
        dwt_write16bitoffsetreg(PLL_CFG_ID, 0, RF_PLL_CFG_CH5);
    

    }

    dwt_write8bitoffsetreg(LDO_RLOAD_ID, 1, LDO_RLOAD_VAL_B1);  
    dwt_write8bitoffsetreg(TX_CTRL_LO_ID, 2, RF_TXCTRL_LO_B2);
  
    dwt_write8bitoffsetreg(PLL_CAL_ID, 0, RF_PLL_CFG_LD);        // Extend the lock delay

    //Verify PLL lock bit is cleared
    dwt_write32bitoffsetreg(SYS_STATUS_ID, 0, SYS_STATUS_CP_LOCK_BIT_MASK);

    ///////////////////////
    // auto cal the PLL and change to IDLE_PLL state
    dwt_setdwstate(DWT_DW_IDLE);

    for (flag=1,cnt=0;cnt<MAX_RETRIES_FOR_PLL;cnt++)
    {
       UART_puts("cnt:");         // Add for debug
       Serial.print(cnt, DEC);    // Add for debug
       UART_puts("\r\n");         // Add for debug
       //deca_usleep(DELAY_20uUSec);  /// comment-out
        if ((dwt_read8bitoffsetreg(SYS_STATUS_ID, 0) & SYS_STATUS_CP_LOCK_BIT_MASK))
        {//PLL is locked
      UART_puts("PLL is locked..");
      UART_puts("\r\n");
            flag=0;
            break;
        }
    }
---------------------

GGyul-E, 2024-11-08:
It's because the delay is not enough for PLL locking.
By printing UART Logs, it makes enough delay. So the most easy way is just using while loop for checking `SYS_STATUS_CP_LOCK_BIT_MASK` instead of using for loop.

` 
/*for (flag=1,cnt=0;cnt<MAX_RETRIES_FOR_PLL;cnt++)
{
   //UART_puts("cnt:");         // Add for debug
   //Serial.print(cnt, DEC);    // Add for debug
   //UART_puts("\r\n");         // Add for debug
   deca_usleep(DELAY_100uUSec);  /// comment-out
    if ((dwt_read8bitoffsetreg(SYS_STATUS_ID, 0) & SYS_STATUS_CP_LOCK_BIT_MASK))
    {//PLL is locked
  //UART_puts("PLL is locked..");
  //UART_puts("\r\n");
        flag=0;
        break;
    }
}*/

while(dwt_read8bitoffsetreg(SYS_STATUS_ID, 0) & SYS_STATUS_CP_LOCK_BIT_MASK){
        UART_puts("PLL is locked..\r\n");
        flag = 0;
        break;
    }
`

MitsuyoshiHori, 2024-11-10:
Thanks for your comment.
I see, it is due to the lack of delay for the PLL lock.

I will try to find time to check SYS_STATUS_CP_LOCK_BIT_MASK using a while loop instead of a for loop.
As a side note, I have not tried this, but there was some information about another MAX_RETRIES_FOR_PLL setting that I would like to share.

https://forum.qorvo.com/t/esp32-uwb-dw3000-configuration-fails-for-ch9/12789