Makerfabs/Makerfabs-ESP32-UWB-DW3000 #5: Ranging Example

Harveyko, 2022-12-05:
Hello, sorry I am very new to this. I just got two https://www.makerfabs.com/esp32-uwb-dw3000.html from makerfabs. I try to use the example code provided on https://github.com/Makerfabs/Makerfabs-ESP32-UWB-DW3000 where I upload the initiator and responder code. I am confused on if the initiator is the anchor or tag, or which one should I kept stationary and which one should I move? Also may I ask is there any example on Antenna Delay Calibration, I found some resources for dw1000 but not dw3000. Thank you!

ThomasLengeling, 2022-12-07:
I have the same question, did you find a way to use the anchor-tags thanks!?

Harveyko, 2022-12-08:
Unfortunately not

vladvlasov256, 2023-04-03:
I had to fix the [example](https://how2electronics.com/ranging-localization-with-esp32-uwb-dw3000-module/) to make it work.   

Tag: 
```
#define POLL_TX_TO_RESP_RX_DLY_UUS 1720
#define RESP_RX_TIMEOUT_UUS 250
```

Anchor:
```
#define POLL_RX_TO_RESP_TX_DLY_UUS 2000
```

I used the constants from the AES examples [[1]](https://github.com/Makerfabs/Makerfabs-ESP32-UWB-DW3000/blob/3ea4cb14e4429012ae784a3543e146187acd3e09/Dw3000/examples/ex_06e_AES_ss_twr_initiator/ex_06e_AES_ss_twr_initiator.ino) and [[2]](https://github.com/Makerfabs/Makerfabs-ESP32-UWB-DW3000/blob/3ea4cb14e4429012ae784a3543e146187acd3e09/Dw3000/examples/ex_06f_AES_ss_twr_responder/ex_06f_AES_ss_twr_responder.ino).

JerryIeong, 2023-04-04:
> I had to fix the [example](https://how2electronics.com/ranging-localization-with-esp32-uwb-dw3000-module/) to make it work.
>
> Tag:
>
> ```
> #define POLL_TX_TO_RESP_RX_DLY_UUS 1720
> #define [...]

Thank you so much for your help! It works after correcting the values you mentioned!

Basically, what I did for DW3000 ESP32
1. I used "06a_ss_twr_initator" and "06a_ss_twr_responder"
2. change the related values of tag mentioned by Mr. vladvlasov256 (1720 and 250)
3. 2. change the related values of anchor mentioned by Mr. vladvlasov256 (2000)
4. Other content stays the same
5. Then it works

Again! Thank you for your helps!!