Makerfabs/Makerfabs-ESP32-UWB-DW3000 #6: UWB with Apple's Nearby Interaction example

Maxakipust, 2023-01-22:
I would love to see an example of Apple's Nearby Interaction working with one of these. Or having one interact with an air tag. Would this be possible?

MaxIsMyName, 2023-03-09:
@Makerfabs , are you able to ask Qorvo or to port their libdwt_uwb_driver to allow the board to be used, as described on the product page, to interact with Apple's U1 chip?

goku-ks, 2023-04-25:
@Makerfabs, Unfortunately, the esp32 dw3000 is not compatible with this U1 chip if it is not possible to make it work with Apple devices. the claim that it is compatible with U1 chips is wrong. 

it I only the Qorvo chip that is compatible with Apple U1 chips, but when it is combined with esp32, it is no more compatible with Apple U1 chips.

goku-ks, 2023-05-20:
@doublebyte , I had the same experience as @MaxIsMyName explained. This is a untrue claim by @Makerfabs that the board is compatible with U1 chips. May be the chip from Qorvo is, but in combination with ESP32 it is not as of now.

BhargavVadaliya, 2023-06-20:
@MaxIsMyName Do you have any news or information regarding the release date of Qorvo's SDK or Firmware for Xtensa chips?

robogrow, 2023-07-02:
Does anyone want to write some NI code for the ESP32? 

We might be able to come up with some funding for it.

ivnlabs, 2023-07-02:
@robogrow, I want to write code for the ESP32 DW3000 well-documented code, making the DW3000 TAG connects with more than one ANCHORS and give a correct measurement between all the connected anchors, Does this count?

maa-x, 2023-12-25:
For a much cheaper and better product, you can buy Qorvo's DMW3001CDK. It doesn't have WiFi, but you can just make the CDK sent updates over UART and connect an ESP to that.