Makerfabs/Makerfabs-ESP32-UWB-DW3000 #10: Double sided two-way ranging

JakobOvergaard, 2023-04-03:
Hi,
I'm currently doing my master thesis in computer science and I'm trying to create a system to test the accuracy of UWB ranging. \
I'm trying to perform double sided two-way ranging using the ESP32 Dw3000 from [makerfabs](https://www.makerfabs.com/esp32-uwb-dw3000.html). - Eventually the goal is to perform asymmetric DB TWR with multiple anchors. \
The code base is written in c++ inspired by QORVOS own example [code](https://www.qorvo.com/products/d/da007992) 'ds_twr_responder' and 'ds_twr_initiator'. \
I'm using the Arduino IDE and this library.

However I'm having some issues with the code, as the final message from the initator to the responder gives me an error, which I cannot decode.
When the responder polls for the final message it reads the system status register for errors, which includes checking the following subregisters: \

All RX error masks:

- SYS_STATUS_RXPHE
- SYS_STATUS_RXFCE
- SYS_STATUS_RXFSL
- SYS_STATUS_RXSTO
- SYS_STATUS_ARFE
- SYS_STATUS_CIAERR

User defined RX error masks:

- SYS_STATUS_RXFTO_BIT_MASK
- SYS_STATUS_RXPTO_BIT_MASK

and the RXFCG subregister.

Printing the system status register gives me the following output:
"131312" which is the decimal representation of the hex value 200F0 and the binary representation of 100000000011110000. \
I know it should be possible to decode which subregister the error is coming from, however I have a hard time figuring it out.\

The code for the intiator and responder is attached below.

Does anyone have any idea what the error could be and how to fix it?

Additionally, I'm very interested in getting in contact with anybody having experience with the ESP32 Dw3000 from makerfabs.

---

Initator code:

```c++
#include "dw3000.h"

#define APP_NAME "SS TWR INIT v1.0"

// connection pins
const uint8_t PIN_RST = 27; // reset pin
const uint8_t PIN_IRQ = 34; // irq pin
const uint8_t PIN_SS = 4;   // spi select pin

/* Default communication configuration. We use default non-STS DW mode. */
static dwt_config_t config = {
    5,                /* Channel number. */
    DWT_PLEN_128,     /* Preamble length. Used in TX only. */
    DWT_PAC8,         /* Preamble acquisition chunk size. Used in RX only. */
    9,                /* TX preamble code. Used in TX only. */
    9,                /* RX preamble code. Used in RX only. */
    1,                /* 0 to use standard 8 symbol SFD, 1 to use non-standard 8 symbol, 2 for non-standard 16 symbol SFD and 3 for 4z 8 symbol SDF type */
    DWT_BR_6M8,       /* Data rate. */
    DWT_PHRMODE_STD,  /* PHY header mode. */
    DWT_PHRRATE_STD,  /* PHY header rate. */
    (129 + 8 - 8),    /* SFD timeout (preamble length + 1 + SFD length - PAC size). Used in RX only. */
    DWT_STS_MODE_OFF, /* STS disabled */
    DWT_STS_LEN_64,   /* STS length see allowed values in Enum dwt_sts_lengths_e */
    DWT_PDOA_M0       /* PDOA mode off */
};

/* Inter-ranging delay period, in milliseconds. */
#define RNG_DELAY_MS 1000

/* Default antenna delay values for 64 MHz PRF. See NOTE 2 below. */
[... 487 lines cut ...]
            distance = tof * SPEED_OF_LIGHT;

            test_run_info((unsigned char *)dist_str);
            /* as DS-TWR initiator is waiting for RNG_DELAY_MS before next poll transmission
             * we can add a delay here before RX is re-enabled again */

            Sleep(RNG_DELAY_MS - 10);
          }
        }
        else
        {
          /* Clear RX error events in the DW IC status register. */
          dwt_write32bitreg(SYS_STATUS_ID, SYS_STATUS_ALL_RX_ERR);
        }
      }
    }
  }
  else
  {
    /* Clear RX error events in the DW IC status register. */
    dwt_write32bitreg(SYS_STATUS_ID, SYS_STATUS_ALL_RX_ERR);
  }
}

```

MaxIsMyName, 2023-04-03:
You should probably read the DW3000’s datasheet.

lordxxbyron, 2023-04-12:
I am also interested in collaborating with a double sided two way ranging - you can PM me at lordxxbyron@gmail.com

Gurkengewuerz, 2024-01-17:
Hey @JakobOvergaard and @lordxxbyron did you ever succeeded with a working example?

JakobOvergaard, 2024-02-29:
@Gurkengewuerz Unfortunately not. I ended up doing singled sided two way ranging from one target to multiple anchors, which worked decently, however it had troubles estimating the position of fast moving targets accurately.

We1chJ, 2025-12-04:
@JakobOvergaard Hi! I am doing something very similar to yours, where I have 3 fixed anchors and one tag that is held by a person in hand. I am directly using the example code from this repo, but I am getting inconsistent and fluctuating readings from the UWB modules. When I do one-to-one tag to anchor testing, the readings also have ~0.5 m fluctuations even when everything is stationary. Would you mind sharing a little more on how well does your product work? What is the offset range you are achieving? Any chance you can share your code or any tips? Thanks a lot!